Yaskawa-Global/motoros2 #385: Consider implementing a 'stop motion' ROS service

gavanderhoorn, 2025-03-05:
Context: #384 (but perhaps also #41).

`FollowJointTrajectory` action goals can be cancelled, but motion controlled via the `queue_traj_point` service doesn't have such an interface.

In fact, the [current implementation](https://github.com/Yaskawa-Global/motoros2/blob/34bfdaf4c16a1381ffd4e0d03af5d05001f1eea4/src/ServiceStopTrajMode.c#L43-L75) of `stop_traj_mode` will refuse to do anything if the robot is in motion (also documented [here](https://github.com/Yaskawa-Global/motoros2/blob/34bfdaf4c16a1381ffd4e0d03af5d05001f1eea4/doc/ros_api.md#stop_traj_mode)).

In order to allow for more control by clients, a `stop_motion` (or similarly named) service should be added which stops any ongoing motion.

Any implementation should take care not to interfere with the `FollowJointTrajectory` action state machine (ie: regular action goals) in unexpected ways. Either such a `stop_motion` service should not affect active goals (requiring action clients to cancel goals instead) or it should `ABORT` active goals, similarly to how any on-controller error or alarm would (note: not `CANCELLED`, as that should probably be only returned when clients cancel their action goals).

jimmy-mcelwain, 2025-03-27:
I like the idea of a stop motion service. I can spend some time working on this. 

Related to this, it might be too late in MotoROS2's development, but especially if we add a new service I think that it would be good to change the name of the already existing `stop_traj_mode` service. I have seen users confused, thinking that it would stop motion. It confused me at first. Maybe `reset_motion_mode`, `clear_motion_mode`, or something along those lines would be more intuitive. But that would obviously be a breaking change, so we could put it on hold or just not mess with it.

gavanderhoorn, 2025-03-28:
> I can spend some time working on this.

ok. Let's use this ticket to discuss potential designs.

> I think that it would be good to change the name of the already existing `stop_traj_mode` service. I have seen users confused, thinking that it would stop motion

I can understand why it may be confusing, but both the documentation ([ROS API docs](https://github.com/Yaskawa-Global/motoros2/blob/48d71146fb055c6a71d037abb0014d2f3cfa2031/doc/ros_api.md#stop_traj_mode)) as well as the [error returned when the robot is still in motion](https://github.com/Yaskawa-Global/motoros2/blob/48d71146fb055c6a71d037abb0014d2f3cfa2031/src/ServiceStopTrajMode.c#L53) are quite clear about it: you can't switch the mode when the robot is still moving.

Perhaps this is more of an issue with the documentation than the service itself.

> that would obviously be a breaking change, so we could put it on hold or just not mess with it.

we can introduce breaking changes, we'd just have to be clear about it.

Additionally, especially when renaming a ROS graph resource, we can follow a tick-tock model where we ship binaries with support for both the old and the new name, and emit deprecation warnings when the old name is being referenced/used. Later release(s) will then drop support for the old name.

jimmy-mcelwain, 2025-03-28:
> ok. Let's use this ticket to discuss potential designs.

We already have this stop motion function:

https://github.com/Yaskawa-Global/motoros2/blob/48d71146fb055c6a71d037abb0014d2f3cfa2031/src/MotionControl.c#L1178-L1244

Could we just call that in a new service? 

We could use `std_srvs/Trigger` as the message type. Though that would mean we would either have to choose `bKeepJobRunning` `TRUE` or `FALSE` always. If we want the user to be able to choose that, we could do `std_srvs/SetBool` but I guess it wouldn't be very clear what the user is setting. So I would also be fine with making a custom message type that makes it clear what boolean you are setting (`bKeepJobRunning`).

ted-miller, 2025-03-28:
> Could we just call that in a new service?

Sounds good to me.

> we would either have to choose bKeepJobRunning TRUE or FALSE

`TRUE`.
I think the user should be able to send additional motion instructions without 're-enabling' the robot.

jimmy-mcelwain, 2025-04-04:
There has also been discussion of turning off the servo motors altogether (see #319, #320). Is that something we would roll into this, or would that be a different service? I think it might make more sense if it is a different service, but I figured I'd bring it up before implementing anything.

ted-miller, 2025-04-07:
I think that `stop_motion` should leave the servos on. But I would be on board with having `stop_traj_mode` disable the servos (it's the opposite of `start_xxx_mode`).

jimmy-mcelwain, 2025-04-09:
> I would be on board with having stop_traj_mode disable the servos (it's the opposite of start_xxx_mode).

I'm okay with that. They'll have to call `start_xxx_mode` afterwards anyways if they want to move, so disabling the servos shouldn't have any negative impact. 

> > we would either have to choose bKeepJobRunning TRUE or FALSE
> 
> TRUE.
> I think the user should be able to send additional motion instructions without 're-enabling' the robot.

I'm okay with that

jimmy-mcelwain, 2025-04-09:
Good point. With a little bit of testing on a GP12, it looks like some robot joints change counts when brakes are applied and removed. In home position, L and U usually change by one pulse count since they have the highest load in home position, and the other joints usually don't change at all. If I put the arm in the absolute worst position that I can, sticking nearly straight forward with some light tooling, I am seeing an average of 3 pulse counts of movement on the L joint, but a maximum of 10 pulse counts if I turn it on and immediately turn it off (unrealistically fast/fast enough that a trajectory couldn't execute anyways).

jimmy-mcelwain, 2025-04-09:
>>> we would either have to choose bKeepJobRunning TRUE or FALSE
>> 
>> TRUE.
>> I think the user should be able to send additional motion instructions without 're-enabling' the robot.
> 
> I'm okay with that

Looking at this again, I can imagine a scenario where `stop_motion` is called (from command line or a separate process/thread) while a process where points are continuously being queued is executing and the robot stops very briefly, then just continues as the next point is queued. Is that what we would want? Is that an unreasonable scenario? Should we assume that the process queueing points would stop before there are any calls to `stop_motion`? Because in the scenario where points are constantly being queued as this `stop_motion` is called, this service implemented this way wouldn't do much, and of course we can't call `/stop_traj_mode` in that scenario either since the queue has to be empty for that to succeed. But then if we set `bKeepJobRunning = FALSE` for `stop_motion`, then it would essentially be `/stop_traj_mode` except for it interrupts motion.

jimmy-mcelwain, 2025-05-19:
Alternatively, we could have a pause/unpause or hold/continue system. Though I don't know what we would do with points submitted before or during the pause. I just don't like the idea that a `stop_motion` call doing point queueing could essentially be "lost" if another point is submitted a few milliseconds later.