Yaskawa-Global/motoros2 #388: Custom inform job field in config file is not properly checked

jimmy-mcelwain, 2025-03-07:
I was messing around with new inform jobs while starting to look into developing new features, and I ran into a problem. 

We have the following in the config file:

https://github.com/Yaskawa-Global/motoros2/blob/34bfdaf4c16a1381ffd4e0d03af5d05001f1eea4/config/motoros2_config.yaml#L272

And we have this in the config file parser:

https://github.com/Yaskawa-Global/motoros2/blob/34bfdaf4c16a1381ffd4e0d03af5d05001f1eea4/src/ConfigFile.c#L120

The difference in names matters, and it means that the config file is ignored, and `allow_custom_inform_job` is always set to false. 

Looking at the history, I don't think that this is a new regression, it's been a problem since at least 0.1.0, but I guess people don't often use custom inform jobs. 

Really simple fix. We just need to change the name in the config file allow_custom_inform -> `allow_custom_inform_job`

gavanderhoorn, 2025-03-07:
Seems like an oversight in the PR review (from before the first public release).

Good find.

Should be easy to fix.

I've updated the title to better reflect what's going on btw.

gavanderhoorn, 2025-03-11:
Would be nice if we could come up with a way to check the fields in & structure of the configuration `.yaml` against what the configuration loader expects / assumes.

(nothing more than a thought at the moment, so not creating an issue to track it)

commit d51694cd (2025-03-11): Correct incorrect name in cfg file (custom inform job)

Fix #388.
closed 2025-03-11 by ted-miller