Yaskawa-Global/motoros2 #396: `Ros_MotionControl_AddToIncQueueProcess(..)`: incorrect assumption about 6 axes (debug log statement)

gavanderhoorn, 2025-03-20:
Current implementation:

https://github.com/Yaskawa-Global/motoros2/blob/48d71146fb055c6a71d037abb0014d2f3cfa2031/src/MotionControl.c#L328-L331

note how this always prints the first 6 values in `ctrlGroup->trajectoryIterator->pos`, regardless of the number of axes in the group.

So far this only looks like an issue in this particular `Ros_Debug_BroadcastMsg(..)` call, but it should be fixed regardless.

gavanderhoorn, 2025-03-20:
Same code in MotoROS1 [here](https://github.com/ros-industrial/motoman/blob/f392352035480df440c37e5543c5d6960017bfc7/motoman_driver/MotoPlus/MotionServer.c#L1771-L1773).

jimmy-mcelwain, 2025-03-28:
> So far this only looks like an issue in this particular Ros_Debug_BroadcastMsg(..) call, but it should be fixed regardless.

I think that it is an issue here too

https://github.com/Yaskawa-Global/motoros2/blob/48d71146fb055c6a71d037abb0014d2f3cfa2031/src/CtrlGroup.c#L189-L199

closed 2025-04-03 by jimmy-mcelwain