Yaskawa-Global/motoros2 #402: Bad floating point cast to integer in time conversion utils

jimmy-mcelwain, 2025-03-31:
To convert from nanoseconds to milliseconds, which are both stored as integer types, we are multiplying by `0.000001` and then casting to an integer. But this means that they are losing precision and then getting inappropriately truncated. For example, `0.2 s`, aka `200000000 ns` gets truncated to `199 ms`, `0.3 s` gets truncated to `299` ms, etc. 

We should just need to change it to integer division instead of floating point multiplication. 

The problem is in at least two spots

https://github.com/Yaskawa-Global/motoros2/blob/48d71146fb055c6a71d037abb0014d2f3cfa2031/src/TimeConversionUtils.h#L15-L18

https://github.com/Yaskawa-Global/motoros2/blob/48d71146fb055c6a71d037abb0014d2f3cfa2031/src/TimeConversionUtils.h#L37-L40

This may only matter if the duration or timestamp falls right on a whole millisecond value.

jimmy-mcelwain, 2025-03-31:
And while we're looking at these functions, would it be preferable to round nanoseconds to the nearest millisecond, or is simple integer division better here?

gavanderhoorn, 2025-04-01:
Good catch.

We should take this opportunity to add (unit) tests for these functions, similar to the other tests we have.

> would it be preferable to round nanoseconds to the nearest millisecond, or is simple integer division better here?

I would avoid rounding (and flooring, etc) as much as possible.

Unless you're talking about in conversions from nanoseconds to milliseconds, then it would perhaps be better to round, instead of simply truncating.

gavanderhoorn, 2025-04-01:
@jimmy-mcelwain: if we can address this soon, we could include it in `0.2.1` -- which otherwise would only be a maintenance release to update versions of two of our dependencies.

jimmy-mcelwain, 2025-04-01:
> Unless you're talking about in conversions from nanoseconds to milliseconds, then it would perhaps be better to round, instead of simply truncating.

I should have been clear, but yeah this is what I meant

jimmy-mcelwain, 2025-04-01:
> We should take this opportunity to add (unit) tests for these functions, similar to the other tests we have.

Also I think this is a good idea, I will add this when I make a PR for this change. 

> Edit: and we might also want to deal with overflow, although I'm not sure how to signal that to callers -- without changing the prototype that is.

I'll take a look at this too.

jimmy-mcelwain, 2025-04-02:
I think I will truncate instead of rounding, since that is what is done in `rcutils` ([see here](https://github.com/ros2/rcutils/blob/b4c01b649e11ee9703ca8153900c376e27f6eded/include/rcutils/time.h#L31-L43))

I might also just use those macros. I wasn't aware of them until now.

PR #411 (2025-04-16): Time utils fix
Resolves #402

This PR gets rid of the floating point math error that results in inappropriate truncation of ms timestamps. 

It also changes the time utils to work for negative times and negative durations. 

I added a bunch of unit tests. Note that due the problem discussed in #407, the controller will crash if you set the testing flag. So make sure to comment out [these lines](https://github.com/Yaskawa-Global/motoros2/blob/ec13454105c21c9397449f9aa1491a224147cc14/src/main.c#L75-L78) before setting that flag and building. 

This PR is a lot more broad than I was originally pl [... cut]